MelbourneDeveloper/Device.Net #197: Unable to read from the USB Device.

YeetSir, 2021-03-03:
I'm attempting to read a scale via USB with Device.net. I'm able to connect to the scale but I'm unable to read from it. When my code gets to this line:
    await usbDevice.InitializeAsync().ConfigureAwait(false);
 i get "Could not open connection for reading" coming from WindowsHidHandler.cs Line 107.
Can you all please help? :)

﴾─────๑۩۩๑─────﴿My Code﴾─────๑۩۩๑─────﴿

using System;
using System.Linq;
using Device.Net;
using System.Threading.Tasks;
using System.Timers;
using Device.Net.Windows;
using Hid.Net.Windows;
using Usb.Net.Windows;

namespace ScaleCore
{
    public class UsbHidDriver
    {
        //Rice Lake USB Scale Info: VendorId = "vid_1c19" ProductId = "pid_0002"
        public static uint? VendorId { get; set; }
        public static uint? ProductId { get; set; }
        public static string UsbLabel { get; set; } = @"Rice Lake Scale";
        public static int PollingTime { get; set; } = 1000;
        public static int DeviceCount { get; set; }
        public static string UsbDeviceStatus { get; set; }
        public static string UsbDeviceData { get; set; }
        public static string UsbDeviceError { get; set; }

        public static void GetDataFromUsbDevice(uint vendorId, uint productId, string usbLabel, int pollTimeMs)
        {
            if (vendorId != 0)
            {
                VendorId = vendorId;
            }

            if (productId != 0)
            {
                ProductId = productId;
            }

            if (usbLabel.Length <= 0)
            {
                UsbLabel = usbLabel;
            }

            if (pollTimeMs >= 1)
            {
                PollingTime = pollTimeMs;
            }

            var t = new Timer(PollingTime) {AutoReset = true};
            t.Elapsed += StartPollingTheUsbDevice;
            t.Start();
        }

        private static void StartPollingTheUsbDevice(Object source, ElapsedEventArgs e)
        {
            Task.Run(async () =>
            {
                try
                {
                    await ConnectViaHidOrUsb();
                }
                catch (Exception ex)
                {
                    Console.WriteLine(Environment.NewLine + ex.Message);
                }
            });
[... 129 lines cut ...]
Logitech - USB Receiver (Hid - 4d1e55b2-f16f-11cf-88cb-001111000030)
Device Path: \\?\hid#vid_046d&pid_c52b&mi_02&col02#7&1d88479f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Vendor: 1133 Product Id: 50475

Logitech - USB Receiver (Hid - 4d1e55b2-f16f-11cf-88cb-001111000030)
Device Path: \\?\hid#vid_046d&pid_c52b&mi_02&col03#7&1d88479f&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
Vendor: 1133 Product Id: 50475

Logitech - USB Receiver (Hid - 4d1e55b2-f16f-11cf-88cb-001111000030)
Device Path: \\?\hid#vid_046d&pid_c52b&mi_01&col02#7&5c4bb5a&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Vendor: 1133 Product Id: 50475

Logitech - USB Receiver (Hid - 4d1e55b2-f16f-11cf-88cb-001111000030)
Device Path: \\?\hid#vid_046d&pid_c52b&mi_00#7&2393a8cf&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Vendor: 1133 Product Id: 50475

Console sample. This sample demonstrates either writing to the first found connected device, or listening for a device and then writing to it. If you listen for the device, you will be able to connect and disconnect multiple times. This represents how users may actually use the device.

1. Write To Connected Device
2. Listen For Device
3. Temperature Monitor (Observer Design Pattern - https://docs.microsoft.com/en-us/dotnet/standard/events/how-to-implement-a-provider#example)

﴾─────๑۩۩๑─────﴿Info﴾─────๑۩۩๑─────﴿
 - Platform: Windows 10
 - Device Type: USB Rice Lake Scale vid_1c19 pid_0002

MelbourneDeveloper, 2021-03-03:
That's the output from the console app. Please post the log. 

Please read

https://melbournedeveloper.github.io/Device.Net/articles/DebuggingLoggingTracing.html

YeetSir, 2021-03-03:
Thank you sir here's what i got:

'Usb.Net.WindowsSample.exe' (CoreCLR: DefaultDomain): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Private.CoreLib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Usb.Net.WindowsSample.dll'. Symbols loaded.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Step into: Stepping over non-user code 'Usb.Net.WindowsSample.Program.Main'
Step into: Stepping over non-user code 'Usb.Net.WindowsSample.Program.<Main>d__4..ctor'
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Collections.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Device.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\netstandard.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.Abstractions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Usb.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Hid.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\SerialPort.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.DependencyInjection.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.ComponentModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.DependencyInjection.Abstractions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Options.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Linq.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.Debug.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Collections.Concurrent.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Resources.ResourceManager.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Diagnostics.Tracing.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Runtime.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Threading.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
[... cut ...]
Device.Net.Windows.WindowsDeviceEnumerator: Debug: The call to SetupDiEnumDeviceInterfaces  returned ERROR_NO_MORE_ITEMS
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Text.Encoding.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Memory.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.

YeetSir, 2021-03-03:
Also here the log with the "Just my Code" option disabled:

'Usb.Net.WindowsSample.exe' (CoreCLR: DefaultDomain): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Private.CoreLib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Usb.Net.WindowsSample.dll'. Symbols loaded.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Step into: Stepping over non-user code 'Usb.Net.WindowsSample.Program.Main'
Step into: Stepping over non-user code 'Usb.Net.WindowsSample.Program.<Main>d__4..ctor'
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Collections.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Device.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\netstandard.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.Abstractions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Usb.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Hid.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\SerialPort.Net.dll'. 
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.DependencyInjection.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.ComponentModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.DependencyInjection.Abstractions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Options.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Linq.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Users\david\Downloads\Device.Net-main\src\Usb.Net.WindowsSample\bin\Debug\net5.0\Microsoft.Extensions.Logging.Debug.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Collections.Concurrent.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Resources.ResourceManager.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Diagnostics.Tracing.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Runtime.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Threading.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
[... cut ...]
Device.Net.Windows.WindowsDeviceEnumerator: Debug: The call to SetupDiEnumDeviceInterfaces  returned ERROR_NO_MORE_ITEMS
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Text.Encoding.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Usb.Net.WindowsSample.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\5.0.2\System.Memory.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.

MelbourneDeveloper, 2021-03-03:
The Windows sample seems to be picking up the device here:

> (Usb - a5dcbf10-6530-11d2-901f-00c04fb951ed)
Device Path: \?\usb#vid_1c19&pid_0002#6&2af6c4ed&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Vendor: 7193 Product Id: 2

Try connecting with the path "\?\usb#vid_1c19&pid_0002#6&2af6c4ed&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}"

Like [this](https://github.com/MelbourneDeveloper/Device.Net/blob/90aa91e584ec2f7e25002c157ad6bd42e87aa618/src/Device.Net.UnitTests/IntegrationTests.cs#L78):

```cs
        public async Task TestConnectToSTMDFUMode_GUID_DEVINTERFACE_USB_DEVICE()
        {
            const string deviceID = @"\?\usb#vid_1c19&pid_0002#6&2af6c4ed&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}";
            var windowsUsbDevice = new UsbDevice(deviceID, new WindowsUsbInterfaceManager(deviceID));
            await windowsUsbDevice.InitializeAsync();
        }
```

YeetSir, 2021-03-04:
Darn, unfortunately, I'm getting this error: "Device handle no good. Error code: 123"

Here is the output log:

Exception thrown: 'Device.Net.Exceptions.ApiException' in Usb.Net.dll
Exception thrown: 'Device.Net.Exceptions.ApiException' in Usb.Net.dll
Exception thrown: 'Device.Net.Exceptions.ApiException' in System.Private.CoreLib.dll
Exception thrown: 'Device.Net.Exceptions.ApiException' in System.Private.CoreLib.dll
Exception thrown: 'Device.Net.Exceptions.ApiException' in System.Private.CoreLib.dll
Exception thrown: 'Device.Net.Exceptions.ApiException' in System.Private.CoreLib.dll

YeetSir, 2021-03-11:
Do you have any other suggestions that I may be able to try? Thank you, good sir!

MelbourneDeveloper, 2021-03-11:
Starting to clutch at straws here. It might be the permissions when trying to connect. 

But, I also think this might be a serial port device. Try running the windows sample and see which com ports show up. When you remove the device, does the com port also disappear? 

Lastly, does the manufacturer have a sample app in some language?

YeetSir, 2021-03-12:
Good Morning! The device that we are attempting to communicate with uses a USB cable. It's Rice Lake Postal Scale Model BP 1214-75S. We can connect, and use this device when we make a .Net Framework project with another USB driver. However, this .net core project using device.net, we cannot communicate with this USB scale. It's it a problem that we are using .net core to talk to a USB device? Here is the manufacturer's product page for this device:  https://www.ricelake.com/en-US/products/product-details/benchpro-series-shipping-postal-digital-scale#/information 

I've reached out the Rice Lake for some sample .net core code and I'm waiting for their response. 

Thank you for all your help here! I can't wait to get this rocking and rolling! :)

MelbourneDeveloper, 2021-03-12:
Ask them if it uses :

Serial port
Usb
Hid

Follow my instructions on checking the com port