MelbourneDeveloper/Device.Net #202: ILoggerFactory copied from Microsoft.Extensions.Logging.Abstraction nuget

oocpawel, 2021-05-02:
I have an error:

Error	CS0433	The type 'ILoggerFactory' exists in both 'Device.Net, Version=4.0.0.0, Culture=neutral, PublicKeyToken=null' and 'Microsoft.Extensions.Logging.Abstractions, Version=5.0.0.0...

I have checked and it looks like this interface has been copied into this repository from Microsoft repository and the worst thing is that its got the same namepsace so i can't use a namespace to point which type should be used. Is there any reason to not include directly Microsoft.Extensions.Logging.Abstraction nuget?

MelbourneDeveloper, 2021-05-02:
That's only true in .net framework 4.5. That's not the case on other platforms

MelbourneDeveloper, 2021-05-02:
Use .net core

oocpawel, 2021-05-02:
Thanks, i have an app in WPF so i need to check if its even posible right now

MelbourneDeveloper, 2021-05-02:
Do your proof of concept on .net core first

If there is a legit problem with 4.5 it can be fixed.