MelbourneDeveloper/Device.Net #203: ILoggingBuilder does not contain a definition for a "AddDebug" and no accessible extension method "AddDebug" accepting a first argument of type "ILoggingBuilder" could be found .....

orcunor, 2021-05-11:
When i try run the example code i get error like this

 var loggerFactory = LoggerFactory.Create((builder) =>
 {
                _ = builder.AddDebug().SetMinimumLevel(LogLevel.Trace);
  });

var hidFactory =
                new FilterDeviceDefinition(vendorId: 0x534C, productId: 0x0001, label: "Trezor One Firmware 1.6.x", usagePage: 65280)
                .CreateWindowsHidDeviceFactory(loggerFactory);

.CreateWindowsHidDeviceFactory does not workin on .Net Framework 5.0 and i try .Net Core 3.1 and still get the same error
Can anybody help me this issue ?

MelbourneDeveloper, 2021-05-11:
Clone the repo and modify the windows sample.

orcunor, 2021-05-15:
How can I do that ? I just started developing software

MelbourneDeveloper, 2021-05-15:
https://melbournedeveloper.github.io/Device.Net/articles/GettingStarted.html#sidetoggle

BryanCrotaz, 2021-06-23:
> https://melbournedeveloper.github.io/Device.Net/articles/GettingStarted.html#sidetoggle

This code doesn't compile in NetCore 2.1 console app with either Device.Net 3.1 or 4.0 from NuGet

closed 2021-07-20 by MelbourneDeveloper