MelbourneDeveloper/Device.Net #204: Can't get Device.Net Samples working in Visual Studio. Errors: Missing XML comment ; Metadata file could not be found

Kablion, 2021-05-16:
https://stackoverflow.com/questions/67561953/cant-get-device-net-samples-working-in-visual-studio-errors-missing-xml-comme

MelbourneDeveloper, 2021-05-16:
@Kablion I can recreate your issue. It seems a recent VS build has changed some code rules, and this has broken the build.

I recommend turning off all code rules right now:
https://docs.microsoft.com/en-us/visualstudio/code-quality/disable-code-analysis?view=vs-2019

As soon as I get time, I will put the build in the pipelines and fix this up.

MelbourneDeveloper, 2021-05-16:
You might need to treat no errors as warnings on all projects:

[image]

Kablion, 2021-05-16:
I could not solve the Issue. 
I tried unticking the Code Analysis Boxes but they are not present. (see the screenshot below)
I also could not find the settings for "Treat warnings as errors".
I then added 
```
<RunAnalyzersDuringBuild>false</RunAnalyzersDuringBuild>
<RunAnalyzersDuringLiveAnalysis>false</RunAnalyzersDuringLiveAnalysis>
<RunAnalyzers>false</RunAnalyzers>
<TreatWarningsAsErrors>false</TreatWarningsAsErrors> 
```
to the Project File of Usb.Net.WindowsSample inside the ` <PropertyGroup>`.
Tried building, but got the same errors. Even after a restart of Visual Studio it did not work.

Do I have to do it on all Projects in the solution? That would be pretty annoying.

Isn't this just a workaround? Just ignoring all errors does not seem right to me. 
But then again I am a total rookie with working in Visual Studio and/or .NET.

[image]

MelbourneDeveloper, 2021-05-16:
@Kablion do some googling on how to turn all the errors off

MelbourneDeveloper, 2021-05-16:
If you figure it out, let me know

MelbourneDeveloper, 2021-05-18:
@Kablion please grab the develop branch. It should compile now.

Kablion, 2021-05-18:
It does work now. But in Visual Studio the Run Button for Usb.Net.WindowsSample only appears after I click on Run Tests.

But aside from this, I realized that this library does not support bluetooth and unfortunatly bluetooth is exactly what I need.
I am trying to read out the battery status of my DS4 controller which is connected via Bluetooth. 
Because of this I will try using HidSharp instead of Device.Net