MelbourneDeveloper/Device.Net #205: Synchronous and asynchronous reading from buffer

orcunor, 2021-05-18:
I want to read buffer both synchronously and asynchronously, but when the asynchronous method is running, I cannot read synchronously. Can anybody help me ?

Here is my sample code:

private async void btnConnect_Click(object sender, EventArgs e)
        {
            try
            {
                //Configure logging
                var loggerFactory = LoggerFactory.Create((builder) =>
                {
                    _ = builder.AddDebug().SetMinimumLevel(LogLevel.Trace);
                });

                

                //Register the factory for creating Hid devices. 
                var hidFactory =
                    new FilterDeviceDefinition(vendorId: 0x1781, productId: 0x7D1)
                    .CreateWindowsHidDeviceFactory(loggerFactory);

                var usbFactory =
                    new FilterDeviceDefinition(vendorId: 0x1781, productId: 0x7D1)
                    .CreateWindowsUsbDeviceFactory(loggerFactory);

                

                //Join the factories together so that it picks up either the Hid or USB device
                var factories = hidFactory.Aggregate(usbFactory);

                //Get connected device definitions
                var deviceDefinitions = (await factories.GetConnectedDeviceDefinitionsAsync().ConfigureAwait(false)).ToList();
                //Console.WriteLine(deviceDefinitions.Count);
               
                if (listBox1.InvokeRequired)
                    listBox1.BeginInvoke((MethodInvoker)delegate () { listBox1.Items.Add("Count of device: " + deviceDefinitions.Count.ToString()); ; });
                     
                else
                    listBox1.Items.Add("Count of device: " + deviceDefinitions.Count.ToString());
               
                 myDevice = await hidFactory.GetDeviceAsync(deviceDefinitions.FirstOrDefault()).ConfigureAwait(false);

                //Initialize the device
                await myDevice.InitializeAsync().ConfigureAwait(false);

                while (myDevice.IsInitialized)
                {
                    buffer = new byte[65];
                    var readBuffer = await myDevice.WriteAndReadAsync(buffer).ConfigureAwait(false);
                    

                    
                    listBox1.Items.Add("Read Async from buffer bytes: " + BitConverter.ToString(readBuffer));
                    
                }
            }
            catch (Exception ex)
            {

                MessageBox.Show(ex.Message);
            }

        }

        private void btnOpen_Click(object sender, EventArgs e)
        {
            try
            {
                var buffer = new byte[9];
                buffer[0] = 0x00;
                buffer[1] = 0x15;
                buffer[2] = 0x00;  
                buffer[3] = 0x00;  
                buffer[4] = 0x01;
                buffer[5] = 0x19;
                buffer[6] = 0x00;
                buffer[7] = 0x00;
                buffer[8] = 0x00;

                //Write and read the data to the device
                var readBuffer = myDevice.WriteAndReadAsync(buffer).ConfigureAwait(false);

               

            }
            catch (Exception ex)
            {

                MessageBox.Show(ex.Message);
            }

        }

When the connect button is pressed, the asynchronous reading continues in the back, but I want to do the synchronous writing and reading operations at the same time.
I am aware that this code is not working correctly. Is there anyone who can help?

MelbourneDeveloper, 2021-05-18:
You need to use the async modifier on the event handler and use the await keyword with WriteAndReadAsync

orcunor, 2021-05-19:
private async void btnOpen_Click(object sender,  EventArgs e)
        {
            try
            {
                var buffer = new byte[9];
                buffer[0] = 0x00;
                buffer[1] = 0x15;
                buffer[2] = 0x00; // 
                buffer[3] = 0x00; // 
                buffer[4] = 0x01;
                buffer[5] = 0x19;
                buffer[6] = 0x00;
                buffer[7] = 0x00;
                buffer[8] = 0x00;

                //Write and read the data to the device
                var readBuffer = await MyDevice.WriteAndReadAsync(buffer).ConfigureAwait(false);
               
            }
            catch (Exception ex)
            {

                MessageBox.Show("Open Click Exception: " + ex.Message);
            }
            

        }

I do code like  that but never changes

MelbourneDeveloper, 2021-05-19:
What's the issue?

orcunor, 2021-05-20:
When the connect button is pressed, I want it to listen to the buffer asynchronously and write the data coming to the listbox, at the same time to read an write it synchronously using the open-close functions and to write the incoming data to the listbox.

This code I wrote listens when the connect button is pressed, but only once writes the data to the listbox and the data from the open-close functions are never written to the listbox. I think it continues to listen asynchronously when the connect button is pressed, when I press the open-close button I think it does not change the thread and I cannot catch the data from the buffer. I can only see the data from the buffer once, and it happens only once when the connect button is clicked.

In short, how can I write and read data from the buffer asynchronously and synchronously at the same time? Is it possible ?

orcunor, 2021-05-20:
When I click the connect button, although there is no action, a data comes from the buffer and it comes only once and I can see it, but I want it to listen continuously for any action and to write the bytes from the buffer into the listbox continuously until the program is closed.

MelbourneDeveloper, 2021-05-29:
> In short, how can I write and read data from the buffer asynchronously and synchronously at the same time? Is it possible 

I'm sorry but I do not understand what you mean. That is not possible. I'm going to close this issue soon.