MelbourneDeveloper/Device.Net #207: How to build under macOS?

mcuee, 2021-06-12:
Just wondering if there is a simple guide to build Device.Net under macOS. Thanks.

I read the help page here but it is not clear how to build Device.Net itself under macOS.
https://melbournedeveloper.github.io/Device.Net/articles/LinuxmacOSSupport.html

mcuee, 2021-06-12:
Restore the project seems to be okay.

```
Device.Net on  main [?] ❯ dotnet restore src/Device.Net.Pipelines.sln 
/Users/mcuee/build/libusb/Device.Net/src/Device.Net.UWP/Device.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Device.Net.UWP/Device.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Hid.Net.UWP/Hid.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Hid.Net.UWP/Hid.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.UWP/Usb.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.UWP/Usb.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android/Usb.Net.Android.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android/Usb.Net.Android.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
  Determining projects to restore...
  All projects are up-to-date for restore.
```

mcuee, 2021-06-12:
Building the solution failed. I am using Mac Mini M1. I can build libusb and libusbdotnet but not Device.Net.

<details>
<summary> Build log -- click to expand </summary>

```
Device.Net on  main [?] took 3s ❯ dotnet build src/Device.Net.Pipelines.sln
Microsoft (R) Build Engine version 16.10.0+4242f381a for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

/Users/mcuee/build/libusb/Device.Net/src/Device.Net.UWP/Device.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Device.Net.UWP/Device.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Hid.Net.UWP/Hid.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Hid.Net.UWP/Hid.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.UWP/Usb.Net.UWP.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.UWP/Usb.Net.UWP.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android/Usb.Net.Android.csproj : warning NU1503: Skipping restore for project '/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android/Usb.Net.Android.csproj'. The project file may be invalid or missing targets required for restore. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net.Pipelines.sln]
  Determining projects to restore...
  All projects are up-to-date for restore.
/Users/mcuee/build/libusb/Device.Net/src/Device.Net.UWP/Device.Net.UWP.csproj(139,3): error MSB4019: The imported project "/usr/local/share/dotnet/sdk/5.0.300/Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the expression in the Import declaration "/usr/local/share/dotnet/sdk/5.0.300//Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" is correct, and that the file exists on disk.
/Users/mcuee/build/libusb/Device.Net/src/Hid.Net.UWP/Hid.Net.UWP.csproj(144,3): error MSB4019: The imported project "/usr/local/share/dotnet/sdk/5.0.300/Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the expression in the Import declaration "/usr/local/share/dotnet/sdk/5.0.300//Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" is correct, and that the file exists on disk.
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.UWP/Usb.Net.UWP.csproj(147,3): error MSB4019: The imported project "/usr/local/share/dotnet/sdk/5.0.300/Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the expression in the Import declaration "/usr/local/share/dotnet/sdk/5.0.300//Microsoft/WindowsXaml/v16.0/Microsoft.Windows.UI.Xaml.CSharp.targets" is correct, and that the file exists on disk.
/Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android/Usb.Net.Android.csproj(89,3): error MSB4019: The imported project "/usr/local/share/dotnet/sdk/5.0.300/Xamarin/Android/Xamarin.Android.CSharp.targets" was not found. Confirm that the expression in the Import declaration "/usr/local/share/dotnet/sdk/5.0.300//Xamarin/Android/Xamarin.Android.CSharp.targets" is correct, and that the file exists on disk.
  Usb.Net.Android.Fakes -> /Users/mcuee/build/libusb/Device.Net/src/Usb.Net.Android.Fakes/bin/Debug/netstandard2.1/Usb.Net.Android.Fakes.dll
/Users/mcuee/.nuget/packages/microsoft.codeanalysis.netanalyzers/5.0.1/build/Microsoft.CodeAnalysis.NetAnalyzers.targets(57,5): warning : The .NET SDK has newer analyzers with version '5.0.2' than what version '5.0.1' of 'Microsoft.CodeAnalysis.NetAnalyzers' package provides. Update or remove this package reference. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
/Users/mcuee/.nuget/packages/microsoft.codeanalysis.netanalyzers/5.0.1/build/Microsoft.CodeAnalysis.NetAnalyzers.targets(57,5): warning : The .NET SDK has newer analyzers with version '5.0.2' than what version '5.0.1' of 'Microsoft.CodeAnalysis.NetAnalyzers' package provides. Update or remove this package reference. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
/Users/mcuee/build/libusb/Device.Net/src/Device.Net/ConnectedDeviceDefinition.cs(23,22): error CS1591: Missing XML comment for publicly visible type or member 'ConnectedDeviceDefinition.ClassGuid' [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
/Users/mcuee/build/libusb/Device.Net/src/Device.Net/ConnectedDeviceDefinition.cs(82,16): error CS1591: Missing XML comment for publicly visible type or member 'ConnectedDeviceDefinition.ConnectedDeviceDefinition(string, DeviceType, uint?, uint?, string, string, string, ushort?, ushort?, ushort?, int?, int?, string, Guid?)' [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
/Users/mcuee/build/libusb/Device.Net/src/Device.Net/ConnectedDeviceDefinition.cs(121,32): error CS1591: Missing XML comment for publicly visible type or member 'ConnectedDeviceDefinition.ToString()' [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
[... cut ...] mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
/Users/mcuee/build/libusb/Device.Net/src/Device.Net/UnsubscribeDisposable.cs(11,21): error CA1816: Change UnsubscribeDisposable.Dispose() to call GC.SuppressFinalize(object). This will prevent derived types that introduce a finalizer from needing to re-implement 'IDisposable' to call it. [/Users/mcuee/build/libusb/Device.Net/src/Device.Net/Device.Net.csproj]
    6 Warning(s)
    292 Error(s)

Time Elapsed 00:00:05.57
```

</details>

MelbourneDeveloper, 2021-06-24:
I'll try it out on macOS at some point. Meantime, try removing the .NET Framework target from all the projects.