MelbourneDeveloper/Device.Net #211: Docs don't specify how to set up factory in Linux

BryanCrotaz, 2021-06-28:
https://melbournedeveloper.github.io/Device.Net/articles/EnumeratingConnectedDevices.html

Only gives example for Windows. As the **CreateWindowsHidDeviceFactory** has the word Windows in it I assume there is an equivalent for Linux, but code completion doesn't discover it.

MelbourneDeveloper, 2021-06-28:
@BryanCrotaz this is the [extension method](https://github.com/MelbourneDeveloper/Device.Net/blob/c0ab331b78b156cf3af91722661c35dae993e31e/src/Device.Net.LibUsb/LibUsbFactoryExtensions.cs#L15) for creating LibUsb device factories. This should work on any platform, but the version of LibUsbDotNet has not been tested on all platforms.

closed 2021-06-28 by MelbourneDeveloper

BryanCrotaz, 2021-06-28:
Can you add this to the documented example?

BryanCrotaz, 2021-06-28:
Do  I also need to call `LibUsbUsbDeviceFactory.Register();` ?

MelbourneDeveloper, 2021-06-28:
> Do I also need to call LibUsbUsbDeviceFactory.Register();

No