MelbourneDeveloper/Device.Net #213: Windows and LibUsb factory extensions don't match

BryanCrotaz, 2021-06-28:
Signatures of Windows and LibUsb extensions don't match:

Windows version will accept `List<FilterDeviceDefinition>` but LibUsb will only accept `ReadOnlyList<FilterDeviceDefinition>`

Suggest changing LibUsb to accept  `IEnumerable<FilterDeviceDefinition> filterDeviceDefinitions` which is backwards compatible.

```
public static IDeviceFactory CreateWindowsUsbDeviceFactory(
        this IEnumerable<FilterDeviceDefinition> filterDeviceDefinitions,
        ILoggerFactory loggerFactory = null,
        GetConnectedDeviceDefinitionsAsync getConnectedDeviceDefinitionsAsync = null,
        GetUsbInterfaceManager getUsbInterfaceManager = null,
        Guid? classGuid = null,
        ushort? readBufferSize = null,
        ushort? writeBufferSize = null
    )

public static IDeviceFactory CreateLibUsbDeviceFactory(
            this IReadOnlyList<FilterDeviceDefinition> filterDeviceDefinitions,
            ILoggerFactory loggerFactory = null,
            int? timeout = null,
            ushort? writeBufferSize = null,
            ushort? readBufferSize = null,
            Func<ConnectedDeviceDefinition, CancellationToken, Task<bool>> supportsDevice = null
            )
```

MelbourneDeveloper, 2021-07-07:
They are different beasts really. It's more that the lib USB one needs a refactor and I couldn't really do that until lib USB dotnet is stable.

MelbourneDeveloper, 2021-07-07:
You shouldn't need to worry about the optional parameters too much.