MelbourneDeveloper/Device.Net #227: Support .NET 4.0

maf-soft, 2021-11-10:
I'm developing with .NET 4.0 because I have to support Windows Server 2003. 

I'm happy that I can use async await anyway.

So maybe it is not much effort to enable device.net for .NET 4.0, too?

MelbourneDeveloper, 2021-11-10:
@maf-soft .net only introduced async await in 4.5. I think that's the minimum.

maf-soft, 2021-11-10:
I'm using it with 4.0 and Visual Studio Community 2019: https://www.nuget.org/packages/Microsoft.Bcl.Async/

MelbourneDeveloper, 2021-11-11:
@maf-soft fair point. 

Are you interested in helping to see if it's possible to target 4.0? 

I can give you a few pointers if you want to try it out. 

If it works, you can submit a PR

maf-soft, 2021-11-11:
Ok, I cannot invest a lot of time, but If you think it's not much effort, I can try with your pointers. Should we talk somewhere else?

MelbourneDeveloper, 2021-11-11:
@maf-soft  I think it will be possible to some extent but there is going to be a lot of fiddly work involved. 

I  changed the csproj file to this as an example:

```xml
<Project Sdk="Microsoft.NET.Sdk">
    <PropertyGroup>
        <TargetFrameworks>net4;</TargetFrameworks>
        <NoWarn>NU5125</NoWarn>
        <DocumentationFile>Device.Net.xml</DocumentationFile>
        <DebugType>embedded</DebugType>
    </PropertyGroup>
    <ItemGroup>
      <Compile Remove="DeviceDataStreamer.cs" />
    </ItemGroup>
    <ItemGroup>
        <None Include="..\..\README.md" Link="README.md" />
        <PackageReference Include="Microsoft.Bcl.Async" Version="1.0.168" />

		<PackageReference Include="Microsoft.CodeAnalysis.NetAnalyzers" Version="5.0.1">
            <PrivateAssets>all</PrivateAssets>
            <IncludeAssets>runtime; build; native; contentfiles; analyzers; buildtransitive</IncludeAssets>
        </PackageReference>
        <PackageReference Include="Microsoft.SourceLink.GitHub" Version="1.0.0">
            <PrivateAssets>all</PrivateAssets>
            <IncludeAssets>runtime; build; native; contentfiles; analyzers; buildtransitive</IncludeAssets>
        </PackageReference>
    </ItemGroup>
</Project>
```

These are the compilation errors I get. You would need to find more BCL libraries.
[image]

MelbourneDeveloper, 2021-11-11:
@maf-soft 
https://www.infoq.com/news/2011/10/ReadOnly-WInRT/

MelbourneDeveloper, 2021-11-11:
@maf-soft if you want to, you can create your own equivalent interface and list implementations...

maf-soft, 2021-11-11:
Thanks for your investigations and the link. Let's see when I find some time to look at it.

maf-soft, 2021-11-14:
According to https://docs.microsoft.com/en-us/dotnet/framework/install/guide-for-developers, .net 4.0, .net 4.5 and even .net 4.5.1 are officially abandoned starting from VS2022.

MelbourneDeveloper, 2021-11-14:
@maf-soft it doesn't matter. If you look at the change I made, you will be able to inject whatever code you like to create the handle. You can continue to use VS 2019

danielchalmers, 2022-06-16:
wanted to note that .NET 4.6.2 is the earliest version that's officially supported by Microsoft as of April 26th: https://devblogs.microsoft.com/dotnet/net-framework-4-5-2-4-6-4-6-1-will-reach-end-of-support-on-april-26-2022

>Customers currently using .NET Framework 4.5.2, 4.6, or 4.6.1 need to update their deployed runtime to a more recent version – at least .NET Framework 4.6.2 before April 26, 2022 – in order to continue to receive updates and technical support.