MelbourneDeveloper/Device.Net #234: The type 'LogLevel' exists in both 'Device.Net and 'Microsoft.Extensions.Logging'

Plutonan, 2021-12-08:
I get this error. 

**Error CS0433**: The type 'LogLevel' exists in both 'Device.Net, Version=4.2.1.0, Culture=neutral, PublicKeyToken=null' and 'Microsoft.Extensions.Logging.Abstractions, Version=6.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'	

[image]

Why are they sharing the same namespace?

**Info**
Platform: Windows C# .net 4.7.2
Device Type: Hid
Version: 4.2.1

NFran30, 2022-02-18:
Is there a way to rectify this while staying with .NET Framework 4.7.2? I am not able use .NET Core

MaverickMartyn, 2022-03-27:
I've run into this issue as well. Any workarounds?
Has this project been abandoned? I don't see any activity in the past 18 months. :/
I am stuck on .NET 4.7 until .NET 7 drops, due to a known framework issue currently being fixed.

vnau, 2023-09-04:
A workaround for this case is to change an alias of the assembly in the project file:

``` xml
<Target Name="ChangeAliasesOfStrongNameAssemblies" BeforeTargets="FindReferenceAssembliesForReferences;ResolveReferences">
    <ItemGroup>
      <ReferencePath Condition="'%(FileName)' == 'Microsoft.Extensions.Logging.Abstractions'">
        <Aliases>MELA</Aliases>
      </ReferencePath>
    </ItemGroup>
  </Target>
```

After that there would be no conflict and you can use aliased types in your code like this:

``` csharp
extern alias MELA;

using MelaLogLevel = MELA::Microsoft.Extensions.Logging.LogLevel;
```