MelbourneDeveloper/Device.Net #236: SerialPort.Net returned device can't be initialized

Xriuk, 2022-04-27:
**Describe the issue**
I'm trying to retrieve a SerialPort device, it gets inside `devices` (`IEnumerable<ConnectedDeviceDefinition>`), as I can see `WindowsSerialPortDeviceFactory` is retrieving it from the Registry (because it has a label assigned), but when I try to read/write to it it says 'The device has not been initialized.'.
Is the device at fault? Because I see that it provides too little info down below, but when I use it with other programs like Realterm to test it, it connects to the COM port and works.
Is there something else I can test? Thanks

**Your Code**
```
using Device.Net;
using Hid.Net.Windows;
using SerialPort.Net.Windows;
using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading.Tasks;
using Usb.Net.Windows;

namespace MyProgram
{
	class Program
	{
		static (int, int)[] vidPids = new[] {
			( 1637 , 20833 ),
			( 11878, 256   ),
			( 11878, 257   ),
			( 11878, 258   ),
			( 11816, 1     ),
			( 4271 , 2     ),
			( 2478 , 12328 )
		};

		static async Task Main(string[] args)
		{
			FilterDeviceDefinition[] devicesDefs = vidPids .Select(vp => new FilterDeviceDefinition(vendorId: (uint)vp.Item1, productId: (uint)vp.Item2)).ToArray();
			var deviceFactory = devicesDefs.CreateWindowsUsbDeviceFactory()
				.Aggregate(devicesDefs.CreateWindowsHidDeviceFactory())
				.Aggregate(new WindowsSerialPortDeviceFactory());
			var devices = await deviceFactory.GetConnectedDeviceDefinitionsAsync();
			var firstConnectedDeviceDefinition = devices.FirstOrDefault();

			IDevice device = await deviceFactory.GetDeviceAsync(firstConnectedDeviceDefinition);
			if (device == null)
				throw new Exception("There were no devices found");

			await device.InitializeAsync();

			TransferResult res = await device.WriteAndReadAsync(Utils.ToCmd("T")); // Here the exception is thrown

			Console.WriteLine(Utils.ToString(res.Data));
		}
	}
}
```

**Log / Stack Trace**
```
info: Device.Net.DeviceFactory[0]
      Getting connected device definitions...
dbug: Device.Net.Windows.WindowsDeviceEnumerator[0]
      About to call SetupDiGetClassDevs for class Guid dee824ef-729b-4a0e-9c14-b7117d33a817. Flags: 18
dbug: Device.Net.Windows.WindowsDeviceEnumerator[0]
      The call to SetupDiEnumDeviceInterfaces  returned ERROR_NO_MORE_ITEMS
info: Device.Net.DeviceFactory[0]
      Getting connected device definitions...
dbug: Device.Net.Windows.WindowsDeviceEnumerator[0]
      About to call SetupDiGetClassDevs for class Guid 4d1e55b2-f16f-11cf-88cb-001111000030. Flags: 18
info: Hid.Net.Windows.WindowsHidApiService[0]
      Calling CreateFile Area: ApiService for DeviceId: \\?\hid#vid_04d9&pid_a09f&mi_01&col01#9&3581eb34&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd. Desired Access: None. Share mode: 3. Creation Disposition: 3
dbug: WindowsHidDeviceFactoryExtensions[0]
      Found device
warn: Hid.Net.Windows.WindowsHidApiService[0]
[... 100 lines cut ...]
      Found device
warn: Hid.Net.Windows.WindowsHidApiService[0]
      Could not get Hid string. Message:  Region: GetHidString Caller: GetSerialNumber
info: Hid.Net.Windows.WindowsHidApiService[0]
      Calling CreateFile Area: ApiService for DeviceId: \\?\hid#vid_0b05&pid_1866&col05#7&289d55ad&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
dbug: WindowsHidDeviceFactoryExtensions[0]
      Found device
warn: Hid.Net.Windows.WindowsHidApiService[0]
      Could not get Hid string. Message:  Region: GetHidString Caller: GetSerialNumber
dbug: Device.Net.Windows.WindowsDeviceEnumerator[0]
      The call to SetupDiEnumDeviceInterfaces  returned ERROR_NO_MORE_ITEMS
info: Device.Net.DeviceFactory[0]
      Checking if the factory supports this connected device. Device Definition: Device Id: \\.\COM6 Label: \Device\ProlificSerial0 Vid:  Pid:
      Read Buffer Size:  Write Buffer Size:
      Manufacturer:  Product Name:
info: Device.Net.DeviceFactory[0]
      Checking if the factory supports this connected device. Device Definition: Device Id: \\.\COM6 Label: \Device\ProlificSerial0 Vid:  Pid:
      Read Buffer Size:  Write Buffer Size:
      Manufacturer:  Product Name:
```

**Info**
 - Platform: Windows
 - Device Type: SerialPort
 - Version: 4.2.1

Xriuk, 2022-04-27:
Also I don't know if this might be a problem, but the device is supposed to be using a baud rate of 2400, I tried instantiating it manually with `WindowsSerialPortDevice` (by providing the port, baud rate, bytes, ...) but it's not helping, it still does not initialize.

Xriuk, 2022-04-27:
I was trying with a RS232 adapter to USB, now I tried with a serial USB cable (on another port of the device) and it's initializing but I get an error during write/read, I guess that it's something inside the Windows ApiService

Xriuk, 2022-04-28:
Just tested out with the default class SerialPort (System.IO.Ports) and it works, is there any reason why you are not using this package in your library?

MelbourneDeveloper, 2022-05-20:
Hi @Xriuk.Thanks for the detailed information. I had hoped to spend more time on serial ports but unfortunately, it was just never a priority because there is so much need for Hid stuff.

I notice here that you are talking to two possible devices: hid and serial. This is exactly what this framework was designed for so so it's a shame that it's not working.

> is there any reason why you are not using this package in your library

I think this would require an extra nuget package. However, you can create your own factory to do this

danielchalmers, 2022-06-16:
> > is there any reason why you are not using this package in your library
> 
> I think this would require an extra nuget package. However, you can create your own factory to do this

Hi, just wanted to add that [`System.IO.Ports`](https://github.com/dotnet/runtime/tree/main/src/libraries/System.IO.Ports) is in the .NET Core BCL [since .NET Core 2.0](https://github.com/dotnet/runtime/issues/14275) and is also [in the regular .NET Framework](https://referencesource.microsoft.com/#System/sys/system/io/ports/SerialPort.cs). It's not a part of [.NET Standard 2.0](https://github.com/dotnet/standard/issues/153) but [does support that standard](https://github.com/dotnet/runtime/blob/2ce8130f4bf4b02393b95719edb0f9d5c6ee1776/src/libraries/System.IO.Ports/src/System.IO.Ports.csproj#L3). 

One option could be to add a .NET Core target alongside [`netstandard2.0;net45`](https://github.com/MelbourneDeveloper/Device.Net/blob/cf17fd4d5e19348e8ff85b20f60184ea4033d111/src/SerialPort.Net/SerialPort.Net.csproj#L3), then reference [the package](https://www.nuget.org/packages/System.IO.Ports) on the .NET Standard target while avoiding the extra package reference on the other targets.