MelbourneDeveloper/Device.Net #239: Send & Receive Data from Usb Hid device

Iveky, 2022-07-03:
I am working on C# hobby Project to poll data from Solar Hybrid Inverter (Axpert clone) over USB (hid) connection and discovered this package on Nuget. 
Communication is ok for short amount data (< 8 bytes) but I don't catch how to receive larger chunks of data:
Device sends everything divided into 8 byte chunks, so, what to do to get 100+ bytes of data, cos this code in example is not enough:
Proper response would be: "(1492932112103803005535;0" but I receive only first part: "(1492932"

Here below  is code that I wrote looking into example.

```
logText("Write command: QSID");
await targetDevice.InitializeAsync();
readBuffer = await targetDevice.WriteAndReadAsync(buffer).ConfigureAwait(false);
if (readBuffer.BytesTransferred < 3)
{
logText("Response from device is too short!");
Console.WriteLine("\r\nResponse from device is too short!\r\n");
return;
}
//readBuffer.Data is byte[]
response = System.Text.Encoding.ASCII.GetString(readBuffer.Data).Replace("\0", "");
logText("Response from device:" + response);
```

Iveky, 2022-07-03:
I think that I found solution for this problem: as device responds with data in chunks of 8 bytes, one need to execute additional Read statements after initial WriteAndRead, for example, this code gives me proper data (after concatenating strings):

```
            readBuffer = await targetDevice.WriteAndReadAsync(buffer).ConfigureAwait(false);
            response = System.Text.Encoding.ASCII.GetString(readBuffer.Data).Replace("\0", "");
            logText("Response from device:" + response);
            readBuffer = await targetDevice.ReadAsync().ConfigureAwait(false);
            response = System.Text.Encoding.ASCII.GetString(readBuffer.Data).Replace("\0", "");
            logText("Response from device:" + response);
            readBuffer = await targetDevice.ReadAsync().ConfigureAwait(false);
            response = System.Text.Encoding.ASCII.GetString(readBuffer.Data).Replace("\0", "");
            logText("Response from device:" + response);
            readBuffer = await targetDevice.ReadAsync().ConfigureAwait(false);
            response = System.Text.Encoding.ASCII.GetString(readBuffer.Data).Replace("\0", "");
            logText("Response from device:" + response);
```

MelbourneDeveloper, 2022-07-06:
You don't need to do that. Specify the read buffer size before initialising the device